wokwi/easyeda2kicad #1: Error in conversion

cmaxwellau, 2019-10-28:
$ easyeda2kicad PCB1.0.json PCB1.0.kicad

Unsupported shape DIMENSION
Unsupported shape DIMENSION
Unsupported shape DIMENSION
Unsupported shape DIMENSION
Unsupported shape HOLE
Unsupported shape HOLE
Unsupported shape HOLE
Unsupported shape HOLE
/usr/local/lib/node_modules/easyeda2kicad/dist/spectra.js:26
            .filter(function (it) { return it !== null; })
             ^

TypeError: Cannot read property 'filter' of undefined
    at encodeObject (/usr/local/lib/node_modules/easyeda2kicad/dist/spectra.js:26:14)
    at encodeValue (/usr/local/lib/node_modules/easyeda2kicad/dist/spectra.js:20:12)
    at Array.map (<anonymous>)
    at encodeObject (/usr/local/lib/node_modules/easyeda2kicad/dist/spectra.js:27:14)
    at encodeValue (/usr/local/lib/node_modules/easyeda2kicad/dist/spectra.js:20:12)
    at Array.map (<anonymous>)
    at encodeObject (/usr/local/lib/node_modules/easyeda2kicad/dist/spectra.js:27:14)
    at encodeValue (/usr/local/lib/node_modules/easyeda2kicad/dist/spectra.js:20:12)
    at Array.map (<anonymous>)
    at encodeObject (/usr/local/lib/node_modules/easyeda2kicad/dist/spectra.js:27:14)

Underlying object:

```
[
  'fp_line',
  [ 'start', -7.549997600000012, 3.599992800000038 ],
  [ 'end', 1.2500101999999913, 3.599992800000038 ],
  [ 'width', 0.049987199999999996 ],
  [ 'layer', undefined ],
  null,
  null
]
```

Original EasyEDA project:

[(https://easyeda.com/sundbyk/gotek)]

urish, 2019-10-28:
Seems like the issue caused by the fact we don't have mapping for layer number 15 ("Mechanical"). I will look into it.

closed 2019-10-29 by urish

urish, 2019-10-29:
I have published a new release (1.1.1) with this fix. I tested it with your board, it seems to convert it successfully. I'd love to have your feedback about the result of the conversion. 

By the way, how did you find out about this project? And how do you find converting easyeda to kicad useful? I'm planning to write a blog post about this tool, and your input can be really useful!

cmaxwellau, 2019-11-01:
Thanks uri,

Will test it out again soon. I found this with a Google search for easyeda,
kicad and conversion. As a node developer by day who is getting back into
some retro computing projects I thought it was worth checking out :)

commit 2ca8fa4e (2019-10-29): fix: add layer 15

Also throw an error when referencing a missing layer.

close #1