wokwi/easyeda2kicad #2: Does able to convert schematic file also?

mekesim, 2019-11-07:
Thank you. It's quite a excellent tool. That can convert easyeda pcb file to kicad, include netlist.
Is there any way to covert PCB file to schematic file? Or directly convert schematic from easyeda?

urish, 2019-11-08:
Thanks for your feedback! Schematic is not currently supported, only the board layout. If there is enough demand, I may add it in the future

yangjuncode, 2019-11-17:
it will be awesome if both schema and pcb can export.

mtiutiu, 2019-11-27:
Awesome indeed so I vote :+1:  for this one also.

mtiutiu, 2020-01-17:
@urish Is there a plan for this one also? It would be helpful in case of big schematics as it gets tedious to be re-created by hand in KiCAD. Indeed PCB conversion is more important at this point but I think this one will be also in the future IMHO. 

I can help with testing of course but not on the development side 'cause I'm not a good programmer :smile:.

Thanks.

urish, 2020-01-18:
@mtiutiu not sure yet - I'm checking with some friend who may want to look into it

urish, 2020-01-30:
@karinchechik started working on this

urish, 2020-02-04:
For the time being, the code only handles the conversions of wires in schematics. The command for converting the schematics is the same as PCB, and the program automatically detects the file type and performs the conversion accordingly:

`easyeda2kicad filename.json filename.sch`

Components, junctions, texts, etc. are not supported yet. Also multi-page schematics are not supported (only the first page will be converted).

urish, 2020-02-20:
The latest release (1.6.0) also add supports for Junctions and No Connect flags in schematics.

thecriticalpoint, 2021-03-21:
Thank you @urish. You made an incredibly useful thing.

Just out of curiosity is what would it take for you to be motivated enough to want to have it convert the easyEDA schematic? If that isn't a valid question in 2021 is there another way to perform this task with possibly another tool that didn't exist in Feb 2020. A lots's happened since then.

urish, 2021-03-21:
@yaybee worked on it in #58, but unfortunately I haven't had the time yet to go over the changes and integrate them

infosisio, 2022-03-16:
I would like to also express my interest for supporting schematic conversion.

gsrobotics, 2022-03-21:
I would like to also express my interest in supporting schematic conversion.

jrsphoto, 2022-03-25:
Yes, please. I know this is an old thread but if you could add schematic conversion you would save me weeks, of time..

urish, 2022-03-26:
Unfortunately, I do not have the bandwidth to add this feature.

yangjuncode, 2022-03-28:
here is a python version: https://github.com/uPesy/easyeda2kicad.py

urish, 2022-03-28:
@yangjuncode thanks for sharing! Cool to see that your version also supports 3d module conversion.

M10CUBE, 2022-03-28:
Guys this is getting better and better. I hope to see soon a final version that does it all. Congratulations to everybody.
Our team without KiCAD will never could do this contribute to the community: www.wizcube.eu
Easyeda is a place where useful material can be found and transferred to  KiCad to easy community designs.
Thanks again guys...

boromyr, 2023-08-13:
Is there a hope that the schematic conversion will be implemented?

KthProg, 2024-01-07:
Really need this. The only reason I'm even using KiCAD is because EasyEDA completely borked the PCB created from my schematic, so I need to restart with the schematic in better software.

channelsixlabs, 2024-02-10:
One more vote for EasyEDA schematic conversion support. Thanks!

nrodday, 2025-01-02:
+1 for the SCH conversion feature :)