wokwi/easyeda2kicad #3: name of the footprint

mekesim, 2019-11-08:
It should be better to use EasyEDA package name according to KiCAD footprint name.

[image]

[image]

urish, 2019-11-11:
Hi,

I'm not sure what you mean, can you try to explain the example in different words?

Thanks!

mekesim, 2019-11-13:
Hi,
Thanks for your reply. And thanks for you develop such great tool, that I can covert the PCB from EasyEDA to KiCAD.

My question is:
in EasyEDA the **footprint** name is **package**, in the above pictures, the switch footprint name is SW-TH_4PIN-L6.00-W6.00. However your tool use ID to name the footprint.
So after convert to KiCAD PCB file, the footprint name is gge1136.

So each part will have a unique name. Even some of them use a same footprint.

Anyway, it still work fine.

urish, 2019-11-13:
I understand now, good point!
I'll have a look at that to see how to fix this.

closed 2019-11-13 by urish

urish, 2019-11-13:
Released version 1.2.0 with this fix

commit a2217631 (2019-11-13): fix: include footprint name

close #3