wokwi/easyeda2kicad #10: No copper zones aren't respected

mtiutiu, 2019-11-26:
Hi,

Seems that after the conversion the **no copper** zones are not taken into consideration and I get a full copper fill for the board. This happens for both sides of the board. See attached pictures and the original project file.

Conversion:
[image]

Original:
[image]

EasyEDA project:
[Touch_Switch_2ch.zip](https://github.com/wokwi/easyeda2kicad/files/3891725/Touch_Switch_2ch.zip)

**easyeda2kicad version: 1.3.1**

urish, 2019-11-26:
Alright, pushed a fix - can you please test and report back?

closed 2019-11-26 by urish

mtiutiu, 2019-11-26:
Wonderful - you're very skilled sir :). Works like a charm now. Thanks again.

Here's the proof:

[image]

urish, 2019-11-26:
Great to see it improving each time :) Thanks for sharing the screenshots

commit 0bffa5f0 (2019-11-26): feat: convert keepout zones

fix #10