wokwi/easyeda2kicad #11: Components package rotation issues

mtiutiu, 2019-11-26:
Hi,

It seems that this **QFN 48** package is rotated **180** degree after conversion. This happens with other SMD components on the board also like some decoupling capacitors around the QFN 48 IC. Please see attached screenshots.

After conversion:
[image]

Original:
[image]

EasyEDA project:
[Touch_Switch_2ch.zip](https://github.com/wokwi/easyeda2kicad/files/3892319/Touch_Switch_2ch.zip)

I tested it using **latest master code** as it was now at this time of writing.

mtiutiu, 2019-11-26:
Other components facing the same issue (180 degree rotation). As seen in the pictures those SMD packages are basically flipped.

After conversion:

[image]

Original:

[image]

urish, 2019-11-27:
So far, it seems to me to rotated components get flipped on the Y-axis, though I haven't managed to figure out a way to determine which ones get flipped

mtiutiu, 2019-12-11:
@urish Hey, any progress on this? I would really like to "port" some of my projects from EasyEDA to KiCAD. Thanks.

urish, 2019-12-11:
Hi @mtiutiu , unfortauntely, I haven't had a chance to work in it since. I'll give it another shot this weekend and keep you posted!

Other than this and the shifting, are there any other issues that prevent the EasyEDA → KiCad conversion?

mtiutiu, 2019-12-11:
> Hi @mtiutiu , unfortauntely, I haven't had a chance to work in it since. I'll give it another shot this weekend and keep you posted!
> 
> Other than this and the shifting, are there any other issues that prevent the EasyEDA → KiCad conversion?

Haven't seen anything else so far.

mtiutiu, 2020-01-08:
Hi @urish any progress with the current issues? Thanks.

urish, 2020-01-09:
Hi @mtiutiu , thanks for the ping - I haven't had a chance to look into it since that weekend.

If you do have some time to recreate a minimal example (with just one or two components) that reproduces the issue that could really be helpful to expedite the process

mtiutiu, 2020-01-16:
Hi @urish 

Sure. Here's another one more simpler:

Original:
[image]

Conversion:
[image]

PCB source and result of conversion:

[sample_project.zip](https://github.com/wokwi/easyeda2kicad/files/4071650/sample_project.zip)

I tested it using latest **master** code as it was now at this time of writing.

Let me know if you need additional details. Thanks.

urish, 2020-01-17:
Thank @mtiutiu !

urish, 2020-01-17:
Alright, I think I found [the issue](https://github.com/wokwi/easyeda2kicad/commit/cd50297a5b37f2a21c8a52d8d5316df80d6c7e9f#diff-e9f7671dae8faa23f88a00a153691f59R56) after some extensive hunt. Can you please try again and report?

mtiutiu, 2020-01-17:
Oh man - you're awesome! This fixed it. I want to try with some other boards that I have and report back. Thanks a lot, I really appreciate it.

mtiutiu, 2020-01-17:
Yep, it seems fine now. I think this ticket can be closed now. Thanks a lot.

urish, 2020-01-17:
Lovely, I will cut a new release

closed 2020-01-17 by urish

urish, 2020-01-17:
Released 1.3.2 with this fix