wokwi/easyeda2kicad #7: Wrong pads orientation for QFN packages

mtiutiu, 2019-11-26:
Hi,

I tried to convert one of my projects to KiCAD using this utility but it generates pads with wrong orientation for **QFN** packages (**pads** seem to be rotated **90 degree**).
See attached picture. I will attach the project file also if it helps.

Thanks for this little nifty tool.

[image]
[Touch_Switch_2ch.zip](https://github.com/wokwi/easyeda2kicad/files/3891540/Touch_Switch_2ch.zip)

urish, 2019-11-26:
Thanks for reporting! Just to confirm, have you used the latest version of the tool, 1.3.1?

You can print the version number by running: easyeda2kicad -v

mtiutiu, 2019-11-26:
Also some vias (in this case a ground one) seems to be shifted as seen in the attached pictures. 

Conversion result:

[image]

Original:
[image]

mtiutiu, 2019-11-26:
> Thanks for reporting! Just to confirm, have you used the latest version of the tool, 1.3.1?
> 
> You can print the version number by running: `easyeda2kicad -v`

Yep - I cloned your repository and used the latest master code:

```
node dist/main.js -v
Version 1.3.1
```

urish, 2019-11-26:
I pushed a fix, can you please check (also for #8)?

closed 2019-11-26 by urish

mtiutiu, 2019-11-26:
Nice - it's fixed now. Thanks. Issue can be closed now along with #8.

Attaching the result of conversion:
[image]

urish, 2019-11-26:
Awesome! I'm looking into the other other issues now

commit a6ef8e3f (2019-11-26): fix: wrong pads orientation for QFN packages

close #7