wokwi/easyeda2kicad #8: Some vias seems to be shifted

mtiutiu, 2019-11-26:
Hi,

I noticed that some vias are shifted (in this case a ground one). Attached is the project file also.

Conversion result:
[image]

Original:
[image]

Project file:
[Touch_Switch_2ch.zip](https://github.com/wokwi/easyeda2kicad/files/3891617/Touch_Switch_2ch.zip)

urish, 2019-11-26:
Are you sure the VIA is shifted? It seems in the correct place to me, just XL1/XL2 pads are rotated? Perhaps this is the same issue as #7

mtiutiu, 2019-11-26:
Sorry, you're right. I was in a rush and took the wrong conclusion. The pads for that crystal are rotated actually and collides with the GND via.

mtiutiu, 2019-11-26:
Indeed this may be related to #7 .

urish, 2019-11-26:
Cool, I think I know where the problem is - seems like the `width` and `height` attributes need to be rotated according to the part/pad orientation when converting between Kicad and EasyEDA. Looking into it right now

mtiutiu, 2019-11-26:
Fixed in #7.

closed 2019-11-26 by mtiutiu