wokwi/easyeda2kicad #9: Some silkscreen elements are shifted and rotated also

mtiutiu, 2019-11-26:
Hi,

Seems that some silkscreen component labels get shifted and rotated during the conversion process. See attached screenshots. Original project attached also.

Converted:
[image]

Original:
[image]

Project:
[Touch_Switch_2ch.zip](https://github.com/wokwi/easyeda2kicad/files/3891652/Touch_Switch_2ch.zip)

**easyeda2kicad version: 1.3.1**

urish, 2019-11-26:
Rotation issue sseems to be similar to #9, I'm not sure yet about why the shifting happens - investigating

mtiutiu, 2019-11-26:
More screenshots to observe the issue (some references are really heavily shifted). 

[image]

[image]

urish, 2019-11-26:
I pushed a fix for the orientation - can you please test now?

shifting hasn't been addressed yet (but should be a bit less noticable with the correct orientation)

mtiutiu, 2019-11-26:
Yep orientation seems to be fixed now. I think some of the shifting issues are caused by the fact that some components are 180 degree rotated as stated in the #11 issue. If that's true then of course the text gets rotated also creating the illusion of being shifted (well...more or less).

Here's the proof about orientation fix and it seems fine now:

[image]

urish, 2020-01-17:
@mtiutiu was this issue fixed as well?

mtiutiu, 2020-01-17:
@urish Yep, seems so.

urish, 2020-01-17:
Great!

closed 2020-01-17 by urish