wokwi/easyeda2kicad #12: Unable to open KiCAD pcb file

mtiutiu, 2020-01-27:
Hi there,

I tried to convert one of my projects to KiCAD and the process seems to be successful (lots of warnings though). When I try to open the generated kicad_pcb file I get this error message and it refuses to go further.

[image]

**KiCAD** version (latest stable): **5.1.5**
**easyeda2kicad** version: **1.3.2**

Source file used for testing:
[custom_pcb.zip](https://github.com/wokwi/easyeda2kicad/files/4116703/custom_pcb.zip)

Thanks.

mtiutiu, 2020-01-29:
@urish 

It seems that the generated kicad pcb file had this entry at the end (I opened the file with an text editor): `(xy NaN NaN)`. 
I deleted that entry and I was able to open the PCB design file. I think it was a keep out zone that I defined in EasyEDA but I'm not sure.
The zone fill doesn't seem to be a grid style as in the original picture. But that is because it's not supported in KiCAD 5.1.5.

mtiutiu, 2020-01-29:
This design is a little bit different as I had to add some overlapping ground fill zones and some floating ground traces over where they intersect. This is because EasyEDA has some bugs when it comes to overlapped ground fill zones or so it seems. So I had to do some ugly "hacks" 😄 . Not ideal I know..but it worked and served to my final purpose.

urish, 2020-01-29:
Yeah, it seems like the keepout zone uses an SVG arc: `M 4367 3248 A 33.8 33.8 0 1 0 4366.99 3248 Z `
I think this is not directly supported by KiCad, so as a workaround, we'll ignore such zones for now.

urish, 2020-01-29:
Regarding the hatched fill (grid style) - yea, unfortunately it's not supported by KiCad :-(

closed 2020-01-29 by urish

mtiutiu, 2020-01-29:
Works as expected now. Thanks.