wokwi/easyeda2kicad #13: Some footprints silkscreen is too large, PCB outline left open

mtiutiu, 2020-01-29:
I just converted one of my designs and some capacitors get a pretty large footprint silkscreen drawing.
Also the board outline seems to be left open. At a first glance the arc seems to be rotated thus breaking up the connection with the enclosing segment/line.

See attached pictures.

Original:
[image]

Conversion result:
[image]

Project files:
[project.zip](https://github.com/wokwi/easyeda2kicad/files/4128328/project.zip)

KiCAD version (latest stable): **5.1.5**
easyeda2kicad version: **1.3.2**

If any other details are needed please let me know.
Thanks.

mtiutiu, 2020-01-29:
Oh..and I forgot to mention that there are two high voltage symbols in the upper part of the board which are being rendered incomplete after the conversion also (the thunder strike arrow is missing).

urish, 2020-01-29:
Capacitor issue fixed in aebe5df.

Board outline arc - seems like the start/end or the angle have to be inverted, but I'm not sure what the logic for determining that should be, so it requires further investigation.

As for the high voltage symbols - these are SOLIDREGIONs of type `solid`, which are not currently supported - can you please open another issue for that?

mtiutiu, 2020-01-30:
@urish Sure. I will open another ticket for the board outline also as it's more ok to be tracked as a separate issue. Thanks.

urish, 2020-01-30:
Thanks! So we can close this one

closed 2020-01-30 by urish