wokwi/easyeda2kicad #15: Some silkscreen symbols are not being rendered completely

mtiutiu, 2020-01-30:
Seems that some symbols present on the PCB silkscreen (top and/or bottom) are not converted completely leading to an incomplete drawing. See the "**high voltage symbols**" in the attached pictures.

Original:
[image]

Conversion result:
[image]

Project files:
[project.zip](https://github.com/wokwi/easyeda2kicad/files/4132711/project.zip)

KiCAD version (latest stable): **5.1.5**
easyeda2kicad version: **1.3.4**

If any other details are needed please let me know.
Thanks.

urish, 2020-02-03:
Can you please check again with latest version from master?

mtiutiu, 2020-02-04:
@urish 

Hi, the conversion process is crashing with an exception now:
```
/home/mtiutiu/Tmp/easyeda2kicad/dist/board.js:319
    var isSmd = shapes.some(function (shape) { return shape[0] === 'pad' && shape[2] === 'smd'; });
                                                           ^

TypeError: Cannot read property '0' of null
    at /home/mtiutiu/Tmp/easyeda2kicad/dist/board.js:319:60
    at Array.some (<anonymous>)
    at convertLib (/home/mtiutiu/Tmp/easyeda2kicad/dist/board.js:319:24)
    at convertShape (/home/mtiutiu/Tmp/easyeda2kicad/dist/board.js:391:21)
    at /home/mtiutiu/Tmp/easyeda2kicad/dist/board.js:403:148
    at Array.map (<anonymous>)
    at Object.convertBoard (/home/mtiutiu/Tmp/easyeda2kicad/dist/board.js:403:118)
    at Object.<anonymous> (/home/mtiutiu/Tmp/easyeda2kicad/dist/main.js:17:84)
```

urish, 2020-02-04:
Should be fixed by #20, can you give it another go?

mtiutiu, 2020-02-04:
Yep. Works now as expected. See attached picture.
[image]

urish, 2020-02-04:
Awesome! released as 1.4.0

closed 2020-02-04 by urish