wokwi/easyeda2kicad #16: PCB outline left open

mtiutiu, 2020-01-30:
Seems that when dealing with an outline that follows an arc path then after conversion the contour is not enclosed properly.
Here I have an example of an arc and a segment that closes the path. At a first glance the arc seems to be rotated thus breaking up the connection with the enclosing segment/line.

See attached pictures.

Original:
[image]

Conversion result:
[image]

Project files:
[project.zip](https://github.com/wokwi/easyeda2kicad/files/4132732/project.zip)

KiCAD version (latest stable): **5.1.5**
easyeda2kicad version: **1.3.4**

If any other details are needed please let me know.
Thanks.

urish, 2020-01-30:
The start/end points of the arc are flipped - we need to figure out a method to determine which is which

urish, 2020-01-31:
Just pushed a fix, should also solve #18. Can you please confirm

mtiutiu, 2020-01-31:
@urish 

Yep, this fix solved both. Although the endpoints of the arc are not perfectly connected with the segment but after a manual editing of the segment it recognized the board outline.

urish, 2020-01-31:
Thanks for checking!

Yes, it seems like KiCad and EasyEDA render the arc slightly different. I suspect this has to do with the fact KiCad only supports one place after the decimal point for angles, so it rounds up the arc angle from 305.175 to 305.2. Unfortunately, if this is indeed the case, I'm afraid we can't do much about it...

closed 2020-01-31 by urish

mtiutiu, 2020-01-31:
@urish 

I thought so. It's hard if not impossible to get each other perfectly aligned. But that's OK I don't mind doing it by hand. Thanks for fixing this also.