wokwi/easyeda2kicad #17: Non GND solid regions are missing after conversion

mtiutiu, 2020-01-30:
**Description:**
If I have in the original design a **solid copper region** created say for a MOSFET cooling and/or for high current paths which are not tied to GND then those aren't being translated.

**Expected result:**
The solid region tied to MOSFET drain pad should be present on both sides of the PCB.

See attached pictures.

Original:
[image]

Another two pictures (top/bottom) for reference:
[image]
[image]

Conversion result:
[image]

Project files:
[project.zip](https://github.com/wokwi/easyeda2kicad/files/4132817/project.zip)

KiCAD version (latest stable): **5.1.5**
easyeda2kicad version: **1.3.4**

If any other details are needed please let me know.
Thanks.

urish, 2020-01-30:
May be related to #15 - needs further investigation

urish, 2020-01-30:
Could be related to #15

mtiutiu, 2020-02-04:
@urish #15 seems to be fixed now and the symbols are rendered correctly. 
The solid region is still missing though.

urish, 2020-02-04:
Right, I'll look into this over the next few days

urish, 2020-02-04:
How about now?

mtiutiu, 2020-02-04:
Yes sir - that's the spirit! Works like magic now. Thanks a lot.

One small issue is present but that's fixable by hand. It seems that the vias inside that solid region even if they have associated the same net name still don't want to connect.

Other than that - **respect!!!**

[image]

Vias appear like not being connected:
[image]

urish, 2020-02-04:
Thanks for the super-quick feedback! 

Yes, I was looking in KiCad, but it seems like they do not support setting the net names for polygons (I even tried [adding a net attribute](https://github.com/wokwi/easyeda2kicad/blob/408b36e34cccd397ab91423305a3e539193b038c/src/board.ts#L452), but it failed so I commented that out). If you do find a way to do this via KiCad UI, let me know and I will look how to implement it.

mtiutiu, 2020-02-04:
@urish I think this ticket can be closed as the initial issue was solved. The rest..well maybe other tickets if it's the case.

closed 2020-02-04 by urish

mtiutiu, 2020-02-04:
@urish can we have a new release including this fix also if not already there? Thanks.

urish, 2020-02-04:
Sure, released as 0.4.1