wokwi/easyeda2kicad #21: Mounting holes are not present after conversion

mtiutiu, 2020-02-05:
**Description:**
Mounting holes are not present after converting a board from EasyEDA to KiCAD.

**Expected result:**
Mounting holes should be present after conversion.

See attached pictures.

Original:
[image]

After conversion:
[image]

Project files:

[project.zip](https://github.com/wokwi/easyeda2kicad/files/4159667/project.zip)

KiCAD version (latest stable): **5.1.5**
easyeda2kicad version: **1.4.1**

If any other details are needed please let me know.
Thanks.

urish, 2020-02-05:
Does KiCad have a tool for creating holes? 

As far as I can tell, there are two indirect ways to achieve this:
1. Add a circle in the Edge.Cuts layer
2. Create a footprint with a pad that has a hole size that equals the size of the pad

But perhaps I missed something?

mtiutiu, 2020-02-05:
@urish No it doesn't have a direct way of doing it. Usually you accomplish this via a designated mounting hole footprint (there is a dedicated library in KiCAD).

urish, 2020-02-05:
@mtiutiu we can't really use the dedicated footprint, as the holes can be of any diameter, so we'd have to either use a circle in the Edge.Cuts layer, or wrap the holes in a footprint. 

Another option would be to open a feature request for KiCad to add support for standalone holes. Do you happen to know some of the developers who work on KiCad? perhaps they can point us at the right direction

mtiutiu, 2020-02-05:
I haven't used KiCAD for some time now but that was the only option I knew of. Maybe things changed meanwhile I'm not sure. The best place to ask for questions is on their main [forum](https://forum.kicad.info) . All the developers listen there.

mtiutiu, 2020-02-06:
@urish Ok I found the block that needs to created in the kicad pcb file source in order to generate a hole.

```
(module MountingHole:MountingHole_1.5mm (layer F.Cu) (tedit 5E3BC186) (tstamp 5E3B09EF)
    (at 109.982 78.232)
    (descr "Mounting Hole 1.5mm, no annular")
    (tags "mounting hole 1.5mm no annular")
    (attr virtual)
    (fp_text reference "" (at 0 0) (layer F.SilkS)
      (effects (font (size 1.27 1.27) (thickness 0.15)))
    )
    (fp_text value "" (at 0 0) (layer F.SilkS)
      (effects (font (size 1.27 1.27) (thickness 0.15)))
    )
    (pad "" np_thru_hole circle (at 0 0) (size 1.5 1.5) (drill 1.5) (layers *.Cu *.Mask))
  )
```
In my case it's a **1.5mm** hole. As I see above everything can be generated pretty easy I guess as it follows a pattern. 

What's left to be determined is the coordinates which is this attribute `(at 109.982 78.232)`. There are two more attributes like `(tedit 5E3BC186)` and `(tstamp 5E3B09EF)` one is the timestamp when it was placed on the board I assume and the other one I can't figure out now what it really means - maybe the last time when it was edited on the board??. Other than that I don't see an issue creating the required fields as they follow the same pattern. 

Did I missed something? Thanks.

mtiutiu, 2020-02-06:
The hole diameter is set here `(pad "" np_thru_hole circle (at 0 0) (size 1.5 1.5) (drill 1.5) (layers *.Cu *.Mask))`. The rest is just metadata so to speak. The most important atrributes are `(size 1.5 1.5) (drill 1.5)`.

So the above module generates a NPTH hole. For plated holes I need to search what it looks like. But I don't remember if EasyEDA has plated holes or if and how can be generated. I'm saying this because the above module block may need to be **generalized** to support both types.

urish, 2020-02-06:
Thanks for looking into this @mtiutiu !

Seems like the holes in EasyEDA are only NPTH:

[image]

[(source)](https://docs.easyeda.com/en/DocumentFormat/3-EasyEDA-PCB-File-Format/index.html#HOLE)

Let me see what I can do with your findings

mtiutiu, 2020-02-06:
@urish 

I just tested with latest master code as it is right now and it seems to work. Yaay!

See here:
[image]

urish, 2020-02-06:
Lovely, I will release shortly. Can we close this issue then?

mtiutiu, 2020-02-06:
> Lovely, I will release shortly. Can we close this issue then?

Yep, seems like this one can be closed now.

closed 2020-02-06 by mtiutiu

urish, 2020-02-06:
Released as part of 1.5.0