wokwi/easyeda2kicad #25: Documentation for building package in npm

coolio986, 2020-03-03:
I was able to npm install -g easyeda2kicad, but when I try to build it I am getting errors. I can run the program without issues, but when building to typescript it does not work.

I have made changes to the conversions of holes to via's and I would like to create a pull request, but since I cannot build, I cannot pull.
`C:\Users\Anthony\AppData\Roaming\npm\node_modules\easyeda2kicad>npm run-script build`

`npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! easyeda2kicad@1.6.0 build: `tsc`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the easyeda2kicad@1.6.0 build script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Anthony\AppData\Roaming\npm-cache\_logs\2020-03-03T21_09_19_260Z-debug.log`

It's possible that I did something wrong.

I was able to import this project into vscode and debug it (to test my converter). It would really be useful for others.

Thanks

urish, 2020-03-03:
Are you able to run the tests? Is there any information in the debug log file that NPM spits (the path to that file is in the last line of the error message)?

coolio986, 2020-03-03:
I ran test:

```
C:\Users\Anthony\AppData\Roaming\npm\node_modules\easyeda2kicad>npm run-script test

> easyeda2kicad@1.6.0 test C:\Users\Anthony\AppData\Roaming\npm\node_modules\easyeda2kicad
> jest

No tests found, exiting with code 1
Run with `--passWithNoTests` to exit with code 0
No files found in C:\Users\Anthony\AppData\Roaming\npm\node_modules\easyeda2kicad.
Make sure Jest's configuration does not exclude this directory.
To set up Jest, make sure a package.json file exists.
Jest Documentation: facebook.github.io/jest/docs/configuration.html
Pattern:  - 0 matches
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! easyeda2kicad@1.6.0 test: `jest`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the easyeda2kicad@1.6.0 test script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Anthony\AppData\Roaming\npm-cache\_logs\2020-03-03T21_48_27_191Z-debug.log
```

```
The log file:

0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'run-script',
1 verbose cli   'test' ]
2 info using npm@6.4.1
3 info using node@v10.15.1
4 verbose run-script [ 'pretest', 'test', 'posttest' ]
5 info lifecycle easyeda2kicad@1.6.0~pretest: easyeda2kicad@1.6.0
6 info lifecycle easyeda2kicad@1.6.0~test: easyeda2kicad@1.6.0
7 verbose lifecycle easyeda2kicad@1.6.0~test: unsafe-perm in lifecycle true
8 verbose lifecycle easyeda2kicad@1.6.0~test: PATH: C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin;C:\Users\Anthony\AppData\Roaming\npm\node_modules\easyeda2kicad\node_modules\.bin;C:\Users\Anthony\AppData\Roaming\npm\node_modules\.bin;C:\Users\Anthony\AppData\Roaming\npm;C:\Program Files\nodejs\;c:\python27\;c:\python27\scripts;c:\programdata\oracle\java\javapath;c:\windows\system32;c:\windows;c:\windows\system32\wbem;c:\windows\system32\windowspowershell\v1.0\;c:\program files (x86)\ati technologies\ati.ace\core-static;c:\users\anthony\.dnx\bin;c:\program files\microsoft dnx\dnvm\;c:\program files\microsoft sql server\130\tools\binn\;c:\program files\git\cmd;c:\program files\geth;c:\program files\crucial\crucial storage executive;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files (x86)\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\Users\Anthony\Downloads\OpenOCD-20190426-0.10.0\bin;C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin;C:\MinGW\msys\1.0\bin;C:\Users\Anthony\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Microsoft VS Code\bin;C:\Users\Anthony\AppData\Local\GitHubDesktop\bin;C:\Users\Anthony\AppData\Local\Microsoft\WindowsApps;C:\Users\Anthony\AppData\Roaming\npm;C:\Users\Anthony\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\EaseUS\Todo Backup\bin\x64\
9 verbose lifecycle easyeda2kicad@1.6.0~test: CWD: C:\Users\Anthony\AppData\Roaming\npm\node_modules\easyeda2kicad
10 silly lifecycle easyeda2kicad@1.6.0~test: Args: [ '/d /s /c', 'jest' ]
11 silly lifecycle easyeda2kicad@1.6.0~test: Returned: code: 1  signal: null
12 info lifecycle easyeda2kicad@1.6.0~test: Failed to exec test script
13 verbose stack Error: easyeda2kicad@1.6.0 test: `jest`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid easyeda2kicad@1.6.0
15 verbose cwd C:\Users\Anthony\AppData\Roaming\npm\node_modules\easyeda2kicad
16 verbose Windows_NT 10.0.18362
17 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "run-script" "test"
18 verbose node v10.15.1
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error easyeda2kicad@1.6.0 test: `jest`
22 error Exit status 1
23 error Failed at the easyeda2kicad@1.6.0 test script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

```

I am not all to familiar with typescript yet, I know more js than anything

urish, 2020-03-03:
Alright, so I think I know - you are working on the package that was installed through NPM, but this package does not include all the source files, just the runtime files (js files that were generated from typescript).

In order to work on the project, you need to:
1. fork it to your own github account (this is needed so that later you could create a pull-request)
2. git checkout from your fork to you local machine
3. `git -b new-branch-new` to create a new branch for your work

then, you can run `npm install` in the directory where you checked out the code, and from that point on you should have a setup where both `npm start` and `npm test` work properly. 

Can you please try this and report back?

coolio986, 2020-03-03:
Yes, I sure will. Just a few moments

coolio986, 2020-03-03:
By the way, Thanks for making this program. You have literally saved me from having to start over in Kicad!

coolio986, 2020-03-03:
Just checked out and ran npm install followed by npm start:

```
C:\Users\Anthony\Documents\GitHub\easyeda2kicad>npm start

> easyeda2kicad@1.6.0 start C:\Users\Anthony\Documents\GitHub\easyeda2kicad
> ts-node src/main.ts

Usage: C:\Users\Anthony\Documents\GitHub\easyeda2kicad\src\main.ts <input.json> [output.kicad_pcb]
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! easyeda2kicad@1.6.0 start: `ts-node src/main.ts`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the easyeda2kicad@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Anthony\AppData\Roaming\npm-cache\_logs\2020-03-03T22_01_25_559Z-debug.log
```

And the log:

```
0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle easyeda2kicad@1.6.0~prestart: easyeda2kicad@1.6.0
6 info lifecycle easyeda2kicad@1.6.0~start: easyeda2kicad@1.6.0
7 verbose lifecycle easyeda2kicad@1.6.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle easyeda2kicad@1.6.0~start: PATH: C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin;C:\Users\Anthony\Documents\GitHub\easyeda2kicad\node_modules\.bin;C:\Users\Anthony\AppData\Roaming\npm;C:\Program Files\nodejs\;c:\python27\;c:\python27\scripts;c:\programdata\oracle\java\javapath;c:\windows\system32;c:\windows;c:\windows\system32\wbem;c:\windows\system32\windowspowershell\v1.0\;c:\program files (x86)\ati technologies\ati.ace\core-static;c:\users\anthony\.dnx\bin;c:\program files\microsoft dnx\dnvm\;c:\program files\microsoft sql server\130\tools\binn\;c:\program files\git\cmd;c:\program files\geth;c:\program files\crucial\crucial storage executive;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\dotnet\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files (x86)\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\Users\Anthony\Downloads\OpenOCD-20190426-0.10.0\bin;C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin;C:\MinGW\msys\1.0\bin;C:\Users\Anthony\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Microsoft VS Code\bin;C:\Users\Anthony\AppData\Local\GitHubDesktop\bin;C:\Users\Anthony\AppData\Local\Microsoft\WindowsApps;C:\Users\Anthony\AppData\Roaming\npm;C:\Users\Anthony\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\EaseUS\Todo Backup\bin\x64\
9 verbose lifecycle easyeda2kicad@1.6.0~start: CWD: C:\Users\Anthony\Documents\GitHub\easyeda2kicad
10 silly lifecycle easyeda2kicad@1.6.0~start: Args: [ '/d /s /c', 'ts-node src/main.ts' ]
11 silly lifecycle easyeda2kicad@1.6.0~start: Returned: code: 1  signal: null
12 info lifecycle easyeda2kicad@1.6.0~start: Failed to exec start script
13 verbose stack Error: easyeda2kicad@1.6.0 start: `ts-node src/main.ts`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid easyeda2kicad@1.6.0
15 verbose cwd C:\Users\Anthony\Documents\GitHub\easyeda2kicad
16 verbose Windows_NT 10.0.18362
17 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "start"
18 verbose node v10.15.1
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error easyeda2kicad@1.6.0 start: `ts-node src/main.ts`
22 error Exit status 1
23 error Failed at the easyeda2kicad@1.6.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

Running `npm test` passed

coolio986, 2020-03-03:
I suppose that means I need to pass arguments using start?

coolio986, 2020-03-03:
Just tried it, yep that works

closed 2020-03-03 by coolio986