wokwi/easyeda2kicad #27: Binaries with releases?

rattboi, 2020-03-05:
This tool looks awesome!

I'm able to build it easily, but I have friends that are less software-inclined and the source-only releases are a barrier-to-entry for them using your great tool.

Would you be open to adding releases as part of the Github Actions CI? 
https://dev.to/jochemstoel/bundle-your-node-app-to-a-single-executable-for-windows-linux-and-osx-2c89

If you'd like, I could submit a PR, but I wanted to test the waters first.

urish, 2020-03-05:
Thanks @rattboi!

Well, you don't really need to build it from source, you can easily install it through npm via `npm i -g easyeda2kicad`, and then use it through the command-line. Is that something that can work for your friends? Perhaps we should improve the documentation?

rattboi, 2020-03-05:
Sure, and that's trivial for me, but convincing windows-based folks to install node/npm is a hurdle in itself. Same with python or ruby or whatever. Lots of people aren't running Cygwin, git-bash, or WSL in windows.

urish, 2020-03-05:
I see, how about a web page where you can open an EasyEDA file and save as KiCad?

rattboi, 2020-03-05:
That'd be a great thing too. 

Are you opposed to generating executables in CI, then? As I said, I could write it up, myself. They'd live in the actions bit, not the git history, so It won't be checking in binaries into git.

urish, 2020-03-05:
Not opposed at all! 

Through, I think a web interface will be easier for most users, and also ensure they are always running with the most up to date version

rattboi, 2020-03-05:
I can get on-board with that as well :)

Having multiple options seems like a win as well, though, so I'll see about getting a PR put together still.

urish, 2020-03-05:
Awesome! Meanwhile, I created a repo for the web interface: https://github.com/wokwi/easyeda2kicad-web

expect some initial version soon :)

tianfeng33, 2020-03-05:
I really appreciate this.  I want to open source more of my files but they are stuck in easyeda.

rattboi, 2020-03-06:
That was amazingly fast. Color me impressed.

urish, 2020-04-11:
Closing, as the online version seems to do the job for now

closed 2020-04-11 by urish