wokwi/easyeda2kicad #28: Only a partial conversion

tianfeng33, 2020-03-06:
Hi,  Just reporting on an issues with the conversion.  I seem to be missing pads and other things.  There is a link to the json file on the git hub.  
[image]

[image]

Here is a link to the project

https://easyeda.com/tianfeng/triple-bypass-v2

Thanks for your work

urish, 2020-03-06:
Thanks for reporting!

Can you please also provide a list of the missing pads/other issues that you found so we can look into them?

tianfeng33, 2020-03-06:
Sure there is a lot. 
I would start with it looks like the ground plane didn't convert. No GND pad
All pads are circular instead of square.
 r1-r4,r6,r22,c6,c7,c20 have no pads. 
 r,5,r7,r8,r10r,r18,r20,r27,r29,r30,r35,c4,c5,c12,c13, are missing the top pad. R11,r24,r25,r31,r31,r37,c3,c19 are missing the bottom.  
Several of the 0603 resistors pads also appear to be different sizes.

urish, 2020-03-06:
Yes, I can definitely see these issues now. I attach the source JSON file for reference, and will start opening separate issues for each kind of problem shortly.

[issue-28.zip](https://github.com/wokwi/easyeda2kicad/files/4299273/issue-28.zip)

urish, 2020-03-06:
Most of the issues (all the resistors related issues) seem to be related to conversion of polygon-shaped pads, so I renamed the issue to reflect that.

urish, 2020-03-07:
Hi @tianfeng33, can you please check again with the last version from master?

Most issues should be resolved now, except for the GND PAD, which should eventually be fixed once #26 will be done and merged.

rattboi, 2020-03-07:
I just tried this via the new online converter and it looks to be the same as before. Will the online version track specific releases or HEAD?

urish, 2020-03-07:
@rattboi the online version tracks the releases. I have just released the latest fixes as 1.7.0, so they should now be reflected there as well.

tianfeng33, 2020-03-17:
I took some time with this today and noticed an issue but I don;t know what is causing it.  The problematic pads are converting as custom circular and not rectangle.  They need to be rotated 180 and changed to rectangle to make them match the correct ones.  I am still experiencing issues with some ground pads not attaching to the ground flood even though the net is gnd.

urish, 2020-03-17:
Hi @tianfeng33, have you used the latest version (1.7.0)? If yes, can you please attach a screenshot highlighting the issues?

tianfeng33, 2020-03-17:
[image]  the non labeled pad is normal

[image]

Then changed to this.
[image]
 
The GND issue might be related to my lack of familiarity with Kicad but everything circled in red is gnd net.  
[image]
 but only some of them are reading that attachment and creating a spoke to ground on a fill.

urish, 2020-03-20:
So here is the deal - it seems like these pads are defined as having a custom shape (polygon) in EasyEDA, and as such, we also convert them to custom shape pads in KiCad. It seems like KiCad can't connect pads with custom shape to the ground fill. I will see if we can automatically detect from the list of points that this looks like a rectangle, and emit a rectangle pad instead of a custom-shape pad.

urish, 2020-03-20:
Can you please check now, with version 1.7.1?

tianfeng33, 2020-03-28:
I checked the other day and forgot to update you.  It is all working much better.  Only one small issues left.  The ground fill does not attach to the two pins on the tssop 14 footprint.  I just connected them with traces.  It is a very small thing but I thought I would mention it

urish, 2020-03-29:
Thanks for the update @tianfeng33! The same happens on my end. The reason for this seems to be the minimum width of the Copper Zone, which is set to 0.254mm. 

[image]

I tried changing it to 0.1mm, and it did the trick for all 3 pads:

[image]

closed 2020-03-29 by urish