wokwi/easyeda2kicad #32: error trying to convert footprint

kimkash, 2020-03-16:
Tried to convert this footprint:
https://easyeda.com/modules/Arduino-Pro-Mini-Footprint_1c2d205f5a224e7abd7498291e7a9dd3

Received error:
"Running easyeda2kicad v1.7.0:
Reading file Arduino Pro Mini Footprint.json...
Parsing content...
Error: unsupported file format"

karinchechik, 2020-03-16:
I'll take a look at this issue and send an update soon

karinchechik, 2020-03-21:
@kimkash - The example you attached is a PCB Module which will be converted to PCB in KiCAD. It is now available. The issue was only in the web version. 

@urish - We should keep this issue open to add support for PCB Lib as well, which will be converted to fileName.kicad_mod

opeRaptor, 2020-04-08:
I am still getting this issue on the web version: Error: unsupported file format

urish, 2020-04-08:
@opeRaptor just to confirm, you are trying to convert an EasyEDA Footprint file, right?

opeRaptor, 2020-04-08:
Yes, trying to convert a EasyEDA footprint file.

I also tried to put a footprint alone on a board as a work around but had another as issue described in #34

urish, 2020-09-05:
Yes, that's exactly the purpose of the issue!

I think that the shape should contain most if not all of the info required for generating the `.kicad_mod` footprint file. TBH I don't remember all the specifics, but @karinchechik looked into it in the past, perhaps she has some additional info that can help

greyltc, 2020-09-05:
I was thinking some info might be missing because I don't see the pin names in that string. I was expecting to see A1...A12 and B1...B12

karinchechik, 2020-09-12:
Currently, any file that is not schematic will be converted to '.kicad_pcb'. @greyltc, if you'll try to convert this PCB (the file you get from the link you attached), you will get a kicad_pcb file. From what I can see, it was fully converted. The A1-A12 and B1-B12 are the PAD numbers. When using a footprint, EasyEDA is showing it as LIB that contains multiple shapes chained to 1 string. In the row of the LIB you can see every PAD with its pad number. So nothing is missing.