wokwi/easyeda2kicad #35: Error: unsupported file format

Seumi, 2020-04-08:
i got a ‘Error: unsupported file format‘ when using the online converter
the json file is exported from easyeda, and the library name is ['EC11E1834403'](https://easyeda.com/component/06da706dc27540619d2a97c512279c2d)
the online converter show
`Running easyeda2kicad v1.7.1:
Reading file SW-TH_EC11E1820402.json...
Parsing content...
Error: unsupported file format
`
also i used the npm version(1.7.1) to conveter the same json file, there is no error but a waring 'Warning: unsupported shape PAD' seem like #34 

this is the origin foorprint on easyeda
[image]
and this is what it be in kicad(convert by npm version)
[image]

urish, 2020-04-08:
Thanks for reporting! #31 and #32 should fix this :-)

Seumi, 2020-04-08:
oh, it seem i also try to convert a footprint like #34 , sorry for my mistake 😔.
Thanks for your outstanding work.😄