wokwi/easyeda2kicad #36: can it export shematic part of easyeda to kicad  too?!

soheilpaper, 2020-04-16:
based of this instruction:

[Introducing: EasyEDA 2 KiCad][1]

by getting [this plugin][2], when i tried to export the file saved by [easyeda][3] :

[![enter image description here][4]][4]

My Problem is in exporting the [Easyeda][3] file to [kicad][5] pcb designing software, i tried to export two pcb and schematic part of easyeda project by 

```

so@so-notebook:~/Desktop/trash/kicad/imported from easyeda/ipt power outler/Project_POWER IOT OUTLET_20200416140429$ easyeda2kicad "1-Schematic_POWER IOT OUTLET.json" board.kicad_pcb
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libcrypto.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libcrypto.so.1.0.0: no version information available (required by node)
/usr/local/lib/node_modules/easyeda2kicad/dist/board.js:524
    var shapes = flatten(input.shape.map(function (shape) { return convertShape(shape, nets); }));
                                     ^

TypeError: Cannot read property 'map' of undefined
    at Object.convertBoard (/usr/local/lib/node_modules/easyeda2kicad/dist/board.js:524:38)
    at Object.<anonymous> (/usr/local/lib/node_modules/easyeda2kicad/dist/main.js:17:84)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:188:16)
    at bootstrap_node.js:609:3
so@so-notebook:~/Desktop/trash/kicad/imported from easyeda/ipt power outler/Project_POWER IOT OUTLET_20200416140429$ ls
'1-PCB_PCB_2020-04-16 03:17:41.json'   README.txt
'1-Schematic_POWER IOT OUTLET.json'
so@so-notebook:~/Desktop/trash/kicad/imported from easyeda/ipt power outler/Project_POWER IOT OUTLET_20200416140429$ easyeda2kicad "1-PCB_PCB_2020-04-16 03:17:41.json" board.kicad_pcb
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libcrypto.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libcrypto.so.1.0.0: no version information available (required by node)
Warning: SOLIDREGION with arcs/circles are not supported yet!

```

so it could convert the PCB part of easyeda project but can not do it for Schematic part of it,a s you can see at above like this:
```
so@so-notebook:~/Desktop/trash/kicad/imported from easyeda/ipt power outler/Project_POWER IOT OUTLET_20200416140429$ easyeda2kicad "1-PCB_PCB_2020-04-16 03:17:41.json" board.kicad_pcb
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libssl.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libcrypto.so.1.0.0: no version information available (required by node)
node: /lib/x86_64-linux-gnu/libcrypto.so.1.0.0: no version information available (required by node)
Warning: SOLIDREGION with arcs/circles are not supported yet!

```

Thanks for your attention.

  [1]: https://medium.com/@urish/introducing-easyeda-2-kicad-37489da1dfa7
  [2]: https://github.com/wokwi/easyeda2kicad
  [3]: https://easyeda.com
  [4]: https://i.stack.imgur.com/NP9mF.png
  [5]: http://www.kicad-pcb.org

urish, 2020-04-16:
Thanks for reporting! Schematics are still work in progress, @karinch is working on adding the missing parts. If you could attach a sample EasyEDA schematic file that fails the conversion, that would be helpful!

soheilpaper, 2020-04-16:
The schematic file added by the main easyeda files, and its links.

urish, 2020-04-16:
Thanks, I'm attaching them to the issue to make it easier for collaborators to find it:

[Project_POWER IOT OUTLET_20200416140429_2.zip](https://github.com/wokwi/easyeda2kicad/files/4489398/Project_POWER.IOT.OUTLET_20200416140429_2.zip)