wokwi/easyeda2kicad #37: file conversation error

svalambhia, 2020-04-22:
Running easyeda2kicad v1.8.0:
Reading file 2.4g telecontrol board_2020-04-23_00-23-43.json...
Parsing content...
Detected EasyEDA board (PCB) file.
Warning: unsupported shape DIMENSION
Warning: unsupported shape DIMENSION
Warning: unsupported shape DIMENSION in footprint gge20f64733a18c5380
Warning: unsupported shape DIMENSION in footprint gge20f64733a18c5380
Warning: unsupported shape DIMENSION in footprint gge20f64733a18c5380
Warning: unsupported shape SVGNODE in footprint gge20f64733a18c5380
Warning: unsupported shape VIA in footprint gge16cbebb8866e1842
Warning: unsupported shape VIA in footprint gge16cbebb8866e1842
Warning: unsupported shape RECT
Warning: unsupported shape RECT
Warning: unsupported shape RECT
Warning: unsupported shape RECT
Warning: unsupported shape RECT
Warning: unsupported shape SVGNODE
Warning: unsupported shape SVGNODE
Warning: unsupported shape SVGNODE
Warning: unsupported shape SVGNODE
Warning: unsupported shape SVGNODE
Writing output to 2.4g telecontrol board_2020-04-23_00-23-43.kicad_pcb...

Conversion failed: InvalidCharacterError: Failed to execute 'btoa' on 'Window': The string to be encoded contains characters outside of the Latin1 range.

urish, 2020-04-22:
Thanks for reporting! Would you mind attaching the board you are trying to convert so we can investigate?

svalambhia, 2020-04-24:
This two items fail. test is very simple file i don't without any text,
2.4g had come text, I thought non english text might have problem so
removed it that also, but it fails after that also.

Best Regards

*Siddharth VALAMBHIA <sid@samupam.com>*

urish, 2020-04-24:
Can you please attach or add a link to a file that fails?

urish, 2020-04-24:
I managed to reproduce the issue using a board that has Hebrew text on it. Working on a fix, should be ready today.

urish, 2020-04-24:
Update: I've uploaded a fix, can you please check again?

svalambhia, 2020-04-30:
> it is not working.

hello sorry for last mail, the files I had sent you are working fine.
Thanks.

Best Regards

*Siddharth VALAMBHIA <sid@samupam.com>*

urish, 2020-04-30:
Excellent, thanks for the update!

closed 2020-04-30 by urish