wokwi/easyeda2kicad #42: Unsupported shape SVGNODE and Unsupported SOLIDREGION

pseudobacon, 2020-10-12:
I tried converting a couple of keyboard PCBs but I got these errors:
[image]

Here are the two projects I tried:
https://lceda.cn/editor#id=47008868b61d4377add794be58e53865
https://lceda.cn/editor#id=04fc83934db54e4f8cd980bc7668f0e2

urish, 2020-10-17:
Attaching the relevant projects as JSON files:

[issue-42.zip](https://github.com/wokwi/easyeda2kicad/files/5395257/issue-42.zip)

urish, 2020-10-17:
Thanks for reporting!

SVGNODE are unlikely to be supported, as the conversion requires a complex process, and they are usually just graphics and do not have a functional use for the PCB.

The `npth` SOLIDREGION requires further investigation. I also opened issues #43 and #44 that cause these files to fail conversion.