wokwi/easyeda2kicad #45: Board outline is not closed

sethmarks, 2020-10-21:
When I convert my EasyEDA board it can be opened in KiCAD just fine.  The outline is as expected but when I view the 3D Viewer I get an error "Board outline is not closed: Unable to find edge with and endpoint of (27.221,-0.444)".

urish, 2020-10-23:
Sometimes that outline does not close due to rounding errors with rounded corners / arcs. Do you use any rounded corners / arcs in your design?

sethmarks, 2020-10-23:
There are internal circles for screw holes but the edges are all lines.  However, many of the lines are at an angle so that could be causing the issues.  It is not a simple design.

urish, 2020-10-23:
Yes, if you can provide the original board JSON file I can have a look to see if this is something that can be fixed