wokwi/easyeda2kicad #48: Unsupported file format

Fshah48, 2020-11-10:
Running easyeda2kicad v1.9.0:
Reading file 1-Schematic_Lora_end_node.json...
Parsing content...
Error: unsupported file format 5

----------------------------------------------------------------------------------------------------------------------
[Project_Lora_end_node_2020-11-10_02-08-58.zip](https://github.com/wokwi/easyeda2kicad/files/5514400/Project_Lora_end_node_2020-11-10_02-08-58.zip)

want to know what i need to do to make it work.
easyeda2kicad did convert the pcb and was loaded into kicad successfully.

urish, 2020-11-10:
Schematic files are not supported at the moment (issue #2 tracks this)

closed 2020-11-10 by urish

bulletmark, 2023-07-26:
It would be better if instead of reporting "Error: unsupported file format 5", it said something like "Error: Schematic files are not supported at the moment".

That would avoid users having to google all over the place then come here and search through the bugs to find the @urish comment above.