wokwi/easyeda2kicad #49: Error: unsupported file format 4

syedkarim, 2020-12-23:
When I upload this file, I receive an error from the website: 
https://wokwi.com/easyeda2kicad?utm_source=wokwi&utm_medium=blog&utm_campaign=easyeda2kicad

I changed the extension from .json to .txt, so it could be attached here. 
[MICRO-USB-SMD_U254-051N-4BH806_2020-12-23_13-34-10.txt](https://github.com/wokwi/easyeda2kicad/files/5736891/MICRO-USB-SMD_U254-051N-4BH806_2020-12-23_13-34-10.txt)

karinchechik, 2020-12-28:
Hi, 

This file is a footprint and footprints should be converted to .kicad_mod - which are not supported yet.
@urish - will it make sense to treat footprints like pcb for the conversion (docType 3)? I'm not familiar with the usage of footprints in kicad.

urish, 2020-12-28:
The `.kicad_mod` files are quite similar to the `.kicad_pcb` format, so it could be as simple as reusing the [logic that converts footprints](https://github.com/wokwi/easyeda2kicad/blob/84a9d4cd8858be9a54df5c65d673b3f9190ad8dc/src/board.ts#L504-L559) and just dumping the result to a `.kicad_mod` file.

Here is an example of a very basic `.kicad_mod` file from one of my projects:

```
(module LOGO (layer F.Cu)
  (at 0 0)
 (fp_text reference "G***" (at 0 0) (layer F.SilkS) hide
  (effects (font (thickness 0.3)))
  )
  (fp_text value "LOGO" (at 0.75 0) (layer F.SilkS) hide
  (effects (font (thickness 0.3)))
  )
)
```

So the output seems very similar to what we already have. 

@karinchechik would you like to take a look at this?

karinchechik, 2020-12-29:
Yes @urish, I will do it in the next few days :)

TheButlah, 2022-02-25:
Did this ever get implemented? EasyEDA has a great selection of footprints, and getting them into kicad would be great :)

urish, 2022-02-25:
Nope :)