wokwi/easyeda2kicad #54: Component Footprint Missing Drill Holes

ceafin, 2021-02-26:
To reproduce:

1. Open LCSC parts catalog and find, for example, component C319164 (micro USB female):
    https://lcsc.com/product-detail/USB-Connectors_XKB-Connectivity-U254-051T-4BH23-S2B_C319164.html/
2. Open EasyEDA model by clicking "EasyEDA Model" button and export the EasyEDA PCB as JSON.
3. Then try to convert this EasyEDA JSON to KiCad PCB using this tool.
4. Open in KiCAD PCBNew. (Looks good)
5. Hit Alt+3 to view 3D model preview, and see how their is no drill holes for the footprint.

Maybe this is expected from the converter tool. Do I need to further edit the footprint to add drill holes?

ceafin, 2021-02-26:
Ah, not just that part. I tried several other components and whole boards, and it seems it doesn't put drill holes for any THT footprints.
So I'm guessing this may just be how it is supposed to work?

yaybee, 2021-02-27:
You probably forgot to switch off the solder paste layer in the 3D viewer, if all pads are gray. You can do this in the 3D Display Options menu. The conversion has no issues as far as I can see.