wokwi/easyeda2kicad #66: 404 - Failed - No File

TEK-Nemesis, 2021-06-21:
Hi.  I've tried with a few different EasyEDA project files and they all look to work but then no file is produced.

Here's a sample log:

Running easyeda2kicad v1.9.2:
Reading file top final_2021-06-21.json...
Parsing content...
Detected EasyEDA board (PCB) file.
Warning: unsupported shape SVGNODE
Warning: unsupported shape SVGNODE
Writing output to top final_2021-06-21.kicad_pcb...
Conversion done!

urish, 2021-06-21:
Can you please attach a sample for a file that fails?

elementzonline, 2021-06-22:
I am also facing the same issue when uploading the EasyEDA json file to the website. On response, it sends back the "No File" 404 response.

I checked with an old JSON file from EasyEDA which was worked in the past, it also has the same issue. 
Anyway, the npm tool is still working fine in the local PC. 

Please see the instructions for running this tool in the npm ecosystem 

I installed npm in my Ubuntu 18.04 with the instructions 
```

sudo apt update
sudo apt install nodejs
sudo apt install npm
```

And finally installed the tool using the following command

`sudo npm install -g easyeda2kicad`

After that I ran the convert command, and worked well
```
easyeda2kicad Downloads/NEW_PCB_2021-06-22.json  output_test.kicad_pcb
```

urish, 2021-06-22:
Thanks for reporting, fixed!

closed 2021-06-22 by urish

TEK-Nemesis, 2021-06-22:
It works perfectly for me now.  Thank you very much!!!