wokwi/easyeda2kicad #67: errer

anjonghyun, 2021-07-07:
[파일 변환 오류 보고서.txt](https://github.com/wokwi/easyeda2kicad/files/6775362/default.txt)
[image]
[image]

urish, 2021-07-07:
Can you please upload / add a link to the board file itself (PCB_SU9_STM32_2021-07-07.json) ?

anjonghyun, 2021-07-07:
[새 폴더.zip](https://github.com/wokwi/easyeda2kicad/files/6775436/default.zip)

anjonghyun, 2021-07-07:
> [시일 성 오와 이스 리시바 .txt](https://github.com/wokwi/easyeda2kicad/files/6775362/default.txt)
> [image]
> [image]

urish, 2021-07-07:
Thanks! I can see this reproduces for me as well.

Looking inside the kicad_pcb file, I can even see what causes the issue:

```
(fp_arc (start NaN NaN) (end -2.05 -1.699) (angle NaN) (width 0.099) (layer "F.SilkS"))
```

I'll investigate where these NaN values from from and fix

closed 2021-07-09 by urish

urish, 2021-07-10:
Fixed, thanks for reporting!

commit 4689d94c (2021-07-09): fix: Error in generated KiCad PCB file #67

Zero-sized arcs result in NaNs in the output file, which KiCad then can't open.

fix #67