wokwi/easyeda2kicad #72: Make a footprint library?

phil-barrett, 2021-11-06:
No problems - it works great. Thanks for your work!

I don't want to design things in easyEDA, I just want to get usable footprints from it.  So, I make a board with the footprint in easyeda, convert it to kicad with your awesome tool and then extract the footprint.  Maybe there's a shortcut way?

urish, 2021-11-06:
You're welcome :) 

I guess it should be possible, but unfortunately I do not have the capacity to look into this. Maybe someone else would?

phil-barrett, 2021-11-06:
To be fair, I usually make my own footprints or use the KiCAD ones.  But unfortunately some datasheets don't have complete footprint data . Connectors DSs often do not say where the center is, for example.  Important for pick and place assembly.  But somehow, easyEDA footprints have that info. Making a small PCB with the device on it to then export and use KiCAD to create a library entry from it isn't totally horrible.