wokwi/easyeda2kicad #96: File will not convert...

AndyW999, 2024-04-03:
Running easyeda2kicad v1.9.5:
Reading file PCBLIB_SA818_2024-04-03(1).json...
Parsing content...
Error: unsupported file format 4

First one ever to have failed!

[PCBLIB_SA818_2024-04-03.json](https://github.com/wokwi/easyeda2kicad/files/14851886/PCBLIB_SA818_2024-04-03.json)

alessandrv, 2024-04-19:
> Running easyeda2kicad v1.9.5: Reading file PCBLIB_SA818_2024-04-03(1).json... Parsing content... Error: unsupported file format 4
> 
> First one ever to have failed!
> 
> [PCBLIB_SA818_2024-04-03.json](https://github.com/wokwi/easyeda2kicad/files/14851886/PCBLIB_SA818_2024-04-03.json)

Just use kicad to import the easyeda file.

Open pcb editor
Import Non-kicad board
Choose easyeda file and import your json

AndyW999, 2024-04-19:
Great!

Thanks

Andy.

closed 2024-04-19 by AndyW999